stereolabs/zed-unity #220: Enabling BatchParameters in Unity freezes

NatureViski, 2021-11-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Enabling batch parameters in sl.dll_ObjectDetectionParameters at start causes Unity to freeze and a memory leak

### Steps to Reproduce

1. Open 3D Object Detection scene
2. Modify ZEDManager.cs startObjectDetection() sl.dll_ObjectDetectionParameters od_param to include od_param.batchParameters.enable = true;
3. Play scene

### Expected Result

Object Detection to start

### Actual Result

Unity Freezes and has a memory leak

### ZED Camera model

ZED2

### Environment

```shell
Windows 10
NVidia GeForce RTX 2080
ZED SDK 3.6.1
Unity Plugin 3.6.0
```

Bvallon-sl, 2021-11-29:
Hi,

The batch option requires to download and optimize an AI model (like any object detection model) at first launch. It is not very well handled in the Unity plugin and it looks like a freeze/crash.
I would recommend to run the object detection bird view sample (with the Batching option enabled : https://github.com/stereolabs/zed-examples/blob/master/object%20detection/birds%20eye%20viewer/cpp/src/main.cpp#L40 ) once. It will download and optimize the model  while printing  without looking like a crash (you have some infos the console explaining you what is going on).

Best regards,
Benjamin Vallon

NatureViski, 2021-11-29:
I have run all the samples and have the AI models in the stereolabs/resources folder and their optimized versions. This is not the problem

closed 2022-04-27 by github-actions[bot]