stereolabs/zed-unity #221: No shadows in build

Pixionrevolution, 2021-12-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When I launch the app in editor the shadows are being displayed "properly" but when I build the app for some reason shadows are not working.
Verified everything regarding shadows was enabled in Quality settings and Graphics and I don't know where to look now.

### Steps to Reproduce

1.Add directional light to scene with ZEDLight.cs 
2.Add object to cast shadow
3.Build

### Expected Result

Shadow displaying in build as "fine" as in editor

### Actual Result

Shadows not displaying only in build

### ZED Camera model

ZED2

### Environment

```shell
Windows
i7
GTX1080

Unity plugin 3.6
```

Pixionrevolution, 2021-12-21:
Does anyone has a solution please?

Bvallon-sl, 2021-12-22:
Hi,

Are you using URP or HDRP ?

I don't have the issue when using the latest plugin version (master branch without URP nor HDRP). Tested with the SimpleMR sample scene using unity2021.1. Note that I increased the shadow strength otherwise it was not very visible. 

Best,
Benjamin V.

Pixionrevolution, 2021-12-22:
Hello,
Thanks for answering me.
I’m not using any specific render pipeline so I basically have the same setup as you do.
What value did you put on the shadow strength ?

Bvallon-sl, 2021-12-22:
I put it at maximum value (1) just to be sure I see it.

Pixionrevolution, 2021-12-22:
Ok, I just fixed my problem by comparing creating a new project and counter check all the settings.
Somehow in Project Settings/Quality, the level were set on default as Very-low.
I don't know why and I feel a bit dumb not to have noticed it. Anyway, the problem is fixed, thanks!

closed 2021-12-22 by Pixionrevolution