stereolabs/zed-unity #226: EntryPointNotFoundException: sl_check_plugin

Nikolatesla-lj, 2022-02-10:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I want to use zed sdk in unity3d, when I import the latest installation package, the error message recurring in unity:

EntryPointNotFoundException: sl_check_plugin
sl.ZEDCamera.CheckPlugin () (at Assets/ZED/SDK/NativeInterface/ZEDCamera.cs:835)
ZEDPluginInspector.UpdateWnd () (at Assets/ZED/Editor/Scripts/ZEDPluginInspector.cs:159)
UnityEditor.EditorApplication.Internal_CallUpdateFunctions () (at /home/bokken/buildslave/unity/build/Editor/Mono/EditorApplication.cs:327)

### Steps to Reproduce

1.Install unity 2020.3.26f1c1
2.Install zed sdk v3.6.4 and cuda
3.Import the Plugin
...
error message appears...

### Expected Result

I hope this installation package does not report errors

### Actual Result

[image]

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu20.04
CPU: intel
GPU: Nvidia gtx 1050
ZED SDK version: v3.6.4
```

Bvallon-sl, 2022-02-10:
Hi,

Thanks for reporting the issue.
I'll look into it right now.

Best regards,
Benjamin Vallon

Bvallon-sl, 2022-02-10:
I updated the package, could you try again ?

Best,
Benjamin V

Nikolatesla-lj, 2022-02-10:
The above error will not occur with the upgraded package, but when I try to run the demo in the Examples folder, unity3d prompts that the camera can be found but cannot be opened, as shown in the following figure
[image]

Nikolatesla-lj, 2022-02-11:
When I open it with ZED explorer, it gets stuck at ''waiting for camera''
However, using matlab the camera can be opened normally.
[image]

Nikolatesla-lj, 2022-02-11:
Ok！
Thank you！
It works！

closed 2022-02-11 by Bvallon-sl