stereolabs/zed-unity #229: plugin v3.7 skeletonscene crash

gksthf2845, 2022-02-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

[image]

other scene  doesn't crash and works well.

confirm ZED Diagnostic.exe
[image]

but, explorer did works wel
[image]
l

### Steps to Reproduce

1. installed cuda 11.5 ,  zed SDK 3.7  
2.run unity project in github zedplugin 3.7 version
3.play example scene at skeleton

(removed old sdk)

### Expected Result

nothing.

just unity proccese crashed.

### Actual Result

unity proccese crashed.

### ZED Camera model

ZED2

### Environment

```shell
window,
geforce 1060
graphic driver latest version.

testted unity version
 2021.1.20f1
 2020.3.18f
 2020.3.15f
 2020.1.10f
```

Bvallon-sl, 2022-02-28:
Hi,

Could you send the diagnostic report at support@stereolabs.com ? 

If possible, the crash report from Unity would be very useful as well. You can find it in :

`C:\Users\USERNAME\AppData\Local\Temp\Unity\Editor\Crashes
`

Note that I pushed a fix this morning on the master branch (some merge conflicts unresolved). you can pull and try again.

Best regards,
Benjamin Vallon

gksthf2845, 2022-03-02:
> t [support@stereolabs.com](mailto:support@stereolabs.com)

can't save diagnostic report. 
After checking the graphic card, immediately process downed.

unity crash report here.

[error.log](https://github.com/stereolabs/zed-unity/files/8166194/error.log)

translation.
잘못된 주소를 액세스하려고 했습니다 >> tried to access the wrong address.

Bvallon-sl, 2022-03-02:
Is the OD model already optimized on your computer ?

If you don't know or is not optimized, can you try to run the body tracking cpp sample (available in the ZED SDK folder, in sample/) ?

Once the model is optimized, comment these lines in the wrapper and try again :

https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs#L2688 
to 

https://github.com/stereolabs/zed-unity/blob/master/ZEDCamera/Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs#L2693

Best regards,
Benjamin Vallon

gksthf2845, 2022-03-03:
sorry , i can't understand mean of OD model .
I don't know what it means even if I search it on Google.
how can i main.cpp build and run?

 i  wrapped what you said and there was no crash.
 but not show ybot model

Bvallon-sl, 2022-03-03:
Sorry I meant [Object Detection Model](https://www.stereolabs.com/docs/api/group__Object__group.html#ga4a212b90726adaaf0fae801b1cc913f4). It's part of our SDK API. When you run the Object detection module, you choose a OD model which allows you to detect different objects in the scene. For example, the **HUMAN_BODY_X" model detects the human skeletons whereas the **MULTI_CLASS_BOX* model detects the several classes of objects and returns their bounding box.

To build and run a cpp program, you have to follow theses steps (link to the documentation [here](https://www.stereolabs.com/docs/app-development/cpp/windows/).
You can try to run a sample and see if it works but we should focus on the error you are getting in the Diagnostic tool as it seems the SDK is not correctly installed. 
Do you have any error message when running the Diagnostic tool during the ZED SDK part (which is red in your image) ?

gksthf2845, 2022-03-07:
thank you, Today Install ZED SDK one more.

done SDK!  diagnostic tool collectly work.

please see this diagnostic log.

{
    "AI Models": {
        "info": [
            "MULTI CLASS DETECTION is not optimized",
            "MULTI CLASS MEDIUM DETECTION is not optimized",
            "MULTI CLASS ACCURATE DETECTION is not optimized",
            "HUMAN BODY FAST DETECTION is not optimized",
            "HUMAN BODY MEDIUM DETECTION is not optimized",
            "PERSON HEAD DETECTION is not optimized",
            "REID ASSOCIATION is not optimized",
            "NEURAL DEPTH is not optimized"
        ],
        "valid": [
            "HUMAN BODY ACCURATE DETECTION is optimized"
        ]
    },
    "Camera Test": {
        "InternalDevicesCount": 2,
        "ZEDCount": 1,
        "camera": {
            "Device ID": 1,
            "Firmware": 1523,
            "Init Output": "OK",
            "Model": "ZED 2",
            "Serial Number": 25329672,
            "valid": [
                "<b>Camera: </b>ZED 2",
                "<b>Firmware: </b>1523"
            ]
        },
        "resolutions": {
            "HD1080@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1080,
                "height (output)": 1080,
                "initialization": "OK",
                "status": true,
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD1080@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 1080,
                "height (output)": 1080,
                "initialization": "OK",
                "status": true,
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD2K@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1242,
                "height (output)": 1242,
                "initialization": "OK",
                "status": true,
                "width (input)": 2208,
                "width (output)": 2208
            },
            "HD720@15": {
                "fps (input)": 15,
                "fps (output)": 15,
[... 355 lines cut ...]
            "cudnn_ops_train64_8.dll",
            "nvinfer.dll",
            "nvinfer_builder_resource.dll",
            "nvinfer_plugin.dll",
            "nvonnxparser.dll",
            "nvparsers.dll",
            "sl_ai64.dll",
            "sl_zed64.dll"
        ],
        "driverVersion": "511.79",
        "drivervDllVersion": "30.0.15.1179",
        "resourcesFiles": [
            "skeleton_accurate_2.0.model",
            "skeleton_accurate_2.0.model_optimized-aaaaw-1-61-11050-8201-8201"
        ],
        "resourcesFilesExpected": [
            "objects_performance_2.1"
        ],
        "runtimeDllVersion": 11050,
        "valid": [
            "<b>ZED SDK version:</b> 3.7.0",
            "<b>CUDA version:</b> V11.5.50"
        ]
    }
}

Bvallon-sl, 2022-03-07:
Hi,

It looks like an usb issue. Can you make sure the camera is connected using a USB 3.0  port ?

gksthf2845, 2022-03-07:
> yes , thanks. 

i know camera doesn't work unless it's usb 3.0.

closed 2022-04-27 by github-actions[bot]

lmbarns, 2022-07-26:
I have a similar issue, the other examples work in Unity but the skeleton demo hangs and then crashes Unity but without an error prompt and no crash log generated. 

Unity 2020.3.13f1 LTS
ZED_SDK_Windows10_cuda11.7_v3.7.6.exe sdk version
ZED_Unity_Plugin_v3.7.1.unitypackage 

The other examples work, and without a crash log it's hard to figure out what's wrong.

jugal-kanousei, 2022-11-24:
Hi there,
did you manage to figure out? I'm having the same issue, Unity Editor just hangs and crashes on play.

Unity 2021.3.4f1 LTS
ZED_SDK_Windows11_cuda11.7 
ZED_Unity_Plugin_v3.8.unitypackage

Bvallon-sl, 2023-01-06:
Hi,

 The AI models need to be downloaded and optimized before the first use.
This process takes a few minutes and is freezing the Unity Editor window until it finishes (this will be fixed in the near future).

Until then, I suggest you to doing this process outside of Unity.
Here is the documentation explaining you how to do it; https://support.stereolabs.com/hc/en-us/articles/9747407795223-How-can-I-optimize-the-ZED-SDK-AI-models-manually-.

Best regards