stereolabs/zed-unity #230: ZED_GreenScreen : How can i render the camera frame in front of everything ?

Gnoknop, 2022-03-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Hello, I am wondering how could I setup the ZED_GreenScreen that the camera frame will be always on the top of virtual objects ?

Almost similar to this post : [Occlusion for real-world frame](https://github.com/stereolabs/zed-unity/issues/169), yet ZED_GreenScreen doesn't use ZED_Forward_Lighting Shader.

Here is what i'm trying to achieve ( left = current state ; right = goal ) :

[image]

Thank you for your help

Bvallon-sl, 2022-03-09:
Hi,

The solution shared in the issues you link should work in the Green Screen shader as well, even if the option is missing.

I added a modified version of the shader which should do what you are looking for.
All you have to do is disable the "Depth occlusion" option in the ZEDManager.cs script.

Best,
Benjamin V.

[GreenScreen.txt](https://github.com/stereolabs/zed-unity/files/8215163/GreenScreen.txt)

Gnoknop, 2022-03-09:
> Hi,
>
> The solution shared in the issues you link should work in the Green Screen shader as well, even if the option is missing.
>
> I added a modified version of the shader which should do what you are [...]

Thanks for answering me. 

I replaced the GreenScreen.shader Code with the modified version you gave me and then disabled the " Depth occlusion " option. **Unfortunately, that method doesn't work.** The Depth occlusion remain active ( despite the option being disable ) and my camera frame is still behind nearest 3D objects.

I noticed something, that might be the cause of the problem ( ? ) :

- Turning off the option " Depth occlusion " on the ZED_GreenScreen Prefab doesn't affect anything at all. The Depth occlusion remain active. ( with the basic shader code and the modified version, same result )

- Whereas turning off the option " Depth occlusion " on the ZED_Rig_Mono Prefab actually disable the Depth occlusion. ( then all 3D objects are rendered in front of the camera frame )

S.D.K Version i use : 3.6.0

Still, i haven't clue how to solve this. I'm looking forward for your help. Thank you

Bvallon-sl, 2022-03-09:
Hi,

If you are using the sdk 3.6.0 you need to do a small change in the shader : 

- change all the occurrence of "NO_DEPTH" with "NO_DEPTH_OCC"

Gnoknop, 2022-03-10:
> * change all the occurrence of "NO_DEPTH" with "NO_DEPTH_OCC"

Work like a charm. Thanks a lot !

closed 2022-04-27 by github-actions[bot]