stereolabs/zed-unity #231: Green Screen Manager : How to dynamically change the " Range " value ?

Gnoknop, 2022-03-10:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Hello, I am wondering how could I dynamically ( in script ) change the " Range " value ?

I tried to change it but, while it does change the " Range " float value and the Range slider, in the inspector, it doesn't affect the final result. The final result only update when you manually use the slider in the inspector.

Is there a way to do it in script ?

S.D.K Version i use : 3.6.0

Thank you for your help

Bvallon-sl, 2022-03-10:
Hi,

You need to call the GreenScreenManager.UpdateShader() function just after modifying the range value (or any other param).

Best,
Benjamin Vallon

Gnoknop, 2022-03-10:
How could i miss it ? Works perfectly, thanks a bunch !

closed 2022-04-27 by github-actions[bot]