stereolabs/zed-unity #232: Keep Receiving Error: "Can't set _ZEDFactorAffectReal value  for Rendering Path VertexLit: only Forward and DeferredShading are supported"

bobsamuel1993, 2022-03-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Was switching from Unity 2019 to Unity 2020 and need to upgrade my ZED as well. When I hit "run" and open the Spatial Mapping menu on the ZED_Manager, the following error occurs: "Can't set _ZEDFactorAffectReal value  for Rendering Path VertexLit: only Forward and DeferredShading are supported". Tried several combinations of SDK/CUDA/ZED Package but keep getting the same error.

- Unity 2020.3.23
- CUDA 11.6
- ZED Package 3.7
- Unity project created with Universal Render Pipeline

### Steps to Reproduce

1. Selected Spatial Mapping scene from Examples
2. Hit play on Unity
3. Opened the Spatial Mapping options on the ZED Manager component
...

### Expected Result

Expected no errors and to be able to begin a spatial mapping scan

### Actual Result

The following error occurs: "Can't set _ZEDFactorAffectReal value  for Rendering Path VertexLit: only Forward and DeferredShading are supported"

### ZED Camera model

ZED Mini

### Environment

```shell
- Windows 10
- Unity 2020.3.23
- CUDA 11.6
- ZED Package 3.7
- Unity project created with Universal Render Pipeline
```

Bvallon-sl, 2022-03-14:
Hi,

First, the ZED SDK is not compatible with Cuda 11.6 yet. I recommend you to install Cuda 11.5 instead even if it's probably not the reason of the issue here.

I tested it myself and was not able to reproduce the error.

Did you follow all the steps to upgrade your project to URP (https://github.com/stereolabs/zed-unity#upgrade-to-urp) ?

Best,
Benjamin Vallon

bobsamuel1993, 2022-03-14:
Benjamin,

Yes I did follow the URP upgrade steps and it didn't make any difference. I also last night tried using CUDA 11.5 with SDK 3.7 and Unity Plugin 3.7, on a different computer from previous, and it came up with the same error. The error occurs whenever I open or close any sub-menu under the ZEDManager script. [](url)

Something else I noticed is that every time I open a project with the plugin, as well as hit play, it keeps popping up the PluginInspector to accept project setting changes. It only disappears for good if I hit "Ignore All"
[image]

-Bobby

Bvallon-sl, 2022-03-15:
Hi,

Can you select the "Left_eye" gameobject in the ZED Stereo prefab and see which renderer is selected (see screenshot)  ?
According to the error, the "wrong" one is selected but I don't see how it's possible.

[image]

closed 2022-04-27 by github-actions[bot]