stereolabs/zed-unity #234: The Model * MULTI_CLASS_BOX_ACCURATE * has not been downloaded/optimized. Use the ZED Diagnostic tool to download/optimze all the AI model you plan to use.

AsIfPu, 2022-04-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I can't run the Object Detection and the SkeletonTracking. Other Examples are normal.I think it's the problem of ai models.

The Model * MULTI_CLASS_BOX_ACCURATE * has not been downloaded/optimized. Use the ZED Diagnostic tool to download/optimze all the AI model you plan to use.
UnityEngine.Debug:LogError (object)
ZEDManager:StartObjectDetection () (at Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs:2691)
ZED3DObjectVisualizer:OnZEDReady () (at Assets/ZED/Examples/Object Detection/Scripts/ZED3DObjectVisualizer.cs:161)
ZEDManager/<InitZED>d__295:MoveNext () (at Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs:1974)
UnityEngine.SetupCoroutine:InvokeMoveNext (System.Collections.IEnumerator,intptr)

Object Detection failed to start. (Error: MODULE_NOT_COMPATIBLE_WITH_CAMERA )
UnityEngine.Debug:Log (object)
ZEDManager/<startObjectDetection>d__319:MoveNext () (at Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs:2775)
UnityEngine.SetupCoroutine:InvokeMoveNext (System.Collections.IEnumerator,intptr)

### Steps to Reproduce

1.Open 3D Object Detection scene
2.Click PlayButton

### Expected Result

[image]

### Actual Result

The camera content can be displayed on the screen, but it is not recognized.

### ZED Camera model

ZED2

### Environment

```shell
Operating System:windows 10 10.0 19043
Processor: Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz (12 CPUs), ~3.2GHz
Memory: 16384MB RAM
Available OS Memory: 16222MB RAM
Page File: 14467MB used, 11483MB available
Windows Dir: C:\Windows
DirectX Version: DirectX 12

Zed Sdk :ZED_SDK_Windows10_cuda11.5_v3.7.1.exe
cuda :cuda_11.5.0_496.13_win10.exe
Zed_Unity:ZED_Unity_Plugin_v3.7.1.unitypackage
```

### Anything else?

{
    "AI Models": {
        "info": [
            "MULTI CLASS DETECTION is not optimized",
            "MULTI CLASS MEDIUM DETECTION is not optimized",
            "MULTI CLASS ACCURATE DETECTION is not optimized",
            "HUMAN BODY FAST DETECTION is not optimized",
            "HUMAN BODY MEDIUM DETECTION is not optimized",
            "HUMAN BODY ACCURATE DETECTION is not optimized",
            "PERSON HEAD DETECTION is not optimized",
            "REID ASSOCIATION is not optimized",
            "NEURAL DEPTH is not optimized"
        ]
    },
[... 266 lines cut ...]
            "cudnn_ops_infer64_8.dll",
            "cudnn_ops_train64_8.dll",
            "nvinfer.dll",
            "nvinfer_builder_resource.dll",
            "nvinfer_plugin.dll",
            "nvonnxparser.dll",
            "nvparsers.dll",
            "sl_ai64.dll",
            "sl_zed64.dll"
        ],
        "driverVersion": "496.13",
        "drivervDllVersion": "30.0.14.9613",
        "resourcesFiles": [
            "objects_performance_2.1.model"
        ],
        "resourcesFilesExpected": [
            "objects_performance_2.1"
        ],
        "runtimeDllVersion": 11050,
        "valid": [
            "<b>ZED SDK version:</b> 3.7.1",
            "<b>CUDA version:</b> V11.5.50"
        ]
    }
}

obraun-sl, 2022-04-05:
Hi,
You need to optimize the model first for your graphic card. You can do it with ZED Diagnostic tool : 
For ex, to optimize multi class accurate : 
$ ZED_Diagnostic.exe -ais 2 

You can also optimize all the models in one command : 
$ ZED_Diagnostic.exe -aio 
(but it takes some time)

AsIfPu, 2022-04-05:
> 您好， 您需要先为您的显卡优化模型。您可以使用 ZED 诊断工具来做到这一点： 例如，优化多类准确： `$ ZED_Diagnostic.exe -ais 2 `
> 
> 您还可以在一个命令中优化所有模型：（ `$ ZED_Diagnostic.exe -aio ` 但这需要一些时间）

Thank you for your reply！AI models has been loaded successfully.
But the 3D Object Detection still doesn't work.

`Object Detection failed to start. (Error: MODULE_NOT_COMPATIBLE_WITH_CAMERA )`

My camera is zed2.

AsIfPu, 2022-04-05:
> Hi, You need to optimize the model first for your graphic card. You can do it with ZED Diagnostic tool : For ex, to optimize multi class accurate : `$ ZED_Diagnostic.exe -ais 2 `
> 
> You can also optimize all the models in one command : `$ ZED_Diagnostic.exe -aio ` (but it takes some time)

Thank you for your reply！AI models has been loaded successfully.
But the 3D Object Detection still doesn't work.

`Object Detection failed to start. (Error: MODULE_NOT_COMPATIBLE_WITH_CAMERA )`

My camera is zed2.

obraun-sl, 2022-04-05:
From the diagnostic file, it seems you are using a ZED not a ZED2
```

"Model": "ZED",
"Serial Number": 12358,
"valid": [
"Camera: ZED",
"Firmware: 1523"
```

AsIfPu, 2022-04-05:
> From the diagnostic file, it seems you are using a ZED not a ZED2
> 
> ```
> 
> "Model": "ZED",
> "Serial Number": 12358,
> "valid": [
> "Camera: ZED",
> "Firmware: 1523"
> ```

Um...I must have taken the wrong camera
Thank you very much for your reply

closed 2022-05-11 by github-actions[bot]