stereolabs/zed-unity #239: Object Detection on SVO file immediately crashes Unity

CivilFX, 2022-08-30:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I've recorded an SVO file from my Zed 2i camera and am attempting to load object detection on the SVO file from Unity. The SVO file plays back correctly. I am attempting to filter only cars but I've tried and the issue happens even when no filters are on.

I've followed this tutorial https://www.stereolabs.com/docs/unity/object-detection/ and have only the ZED Rig and ZED3DObjectVisualizer in the hierarchy. The issue occurs whether I have object detection set to run at start or run it off the ZED Rig editor script. 

I've attached the crash logs. I also exported the SVO file per https://support.stereolabs.com/hc/en-us/articles/360009986754-How-do-I-convert-SVO-files-to-AVI-or-image-depth-sequences-.  However, Github doesn't support AVI and converting it to MP4 leaves a file size too large for Github.

[Editor.log](https://github.com/stereolabs/zed-unity/files/9455405/Editor.log)

### Steps to Reproduce

1. Set up Zed Rig, set input mode to SVO and set asset path.
2. Set up Zed 3D Object Visualizer
3. Hit Play
4. Start Object Detection
...

### Expected Result

Bounding boxes to appear on vehicles. 

### Actual Result

Unity crashes. 

### ZED Camera model

ZED2i

### Environment

```shell
OS: Windows 10 Pro 64-bit
CPU: Intel Core i7-8700
GPU: NVIDIA GeForce GTX 1080
ZED SDK version: 3.7.6
CUDA: 10.2.89
```

### Anything else?

I have not tested object detection on USB mode as I recorded the intersection my laptop using a separate version of Unity and the camera in the field. I need to ensure object detection works on recordings.

Bvallon-sl, 2022-08-31:
Hi,

Can you try to use the c++ body tracking sample and see if you also have the crash ? It is available in your ZED SDK folder or in this[ github repo](https://github.com/stereolabs/zed-examples/tree/master/body%20tracking).

Could you please send us both the SVO and the ZED diagnostic report at support@stereolabs.com ?

Best,
Benjamin Vallon

CivilFX, 2022-09-06:
> Hi,
> 
> Can you try to use the c++ body tracking sample and see if you also have the crash ? It is available in your ZED SDK folder or in this[ github repo](https://github.com/stereolabs/zed-examples/tree/master/body%20tracking).
> 
> Could you please send us both the SVO and the ZED diagnostic report at [support@stereolabs.com](mailto:support@stereolabs.com) ?
> 
> Best, Benjamin Vallon

This solved it. When I opened the body-tracking exe, it downloaded the AI module.

closed 2022-09-06 by CivilFX