stereolabs/zed-unity #244: Import zed plugin to unity crash

muhammadhemdan, 2022-12-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi all,
When I import zed plugin (v3.8.0) for unity (2021.3.15.f1), the unity editor crashes.
I'm using windows10, cuda 10.1, zed sdk 3.4.0 and Nvidia GPU GTX 1660.
How can I solve this problem?
Thanks in advance.

### Steps to Reproduce

1. Create new project from unity editor of type (3D URP).
2. Import zed unity plugin.
3. The editor craches.
...

### Expected Result

I can use the plugin in the project.

### Actual Result

The editor craches.

### ZED Camera model

ZED Mini

### Environment

```shell
zed unity plugin (v3.8.0).
unity (2021.3.15.f1).
windows10, 
cuda 10.1, 
zed sdk 3.4.0,
Nvidia GPU GTX 1660.
```

lsmith24, 2023-01-26:
I am also having this problem, did you find solution?

SLJLMacarit, 2023-02-20:
Hello,
Thanks for reaching!

The Unity plugin v3.8.0 is made for the SDK v3.8. The plugin version is matching with the SDK version, so if you want/need to use SDK v3.4, you need to use plugin v3.4.

However, it should indeed not be crashing, we'll investigate this issue and fix it in a later release.

Best regards,
Jean-Loup Macarit

mikeszabi, 2023-03-06:
Hello,
I have a related problem with Unity 2021.3.20f1. ZED SDK 3.8 is installed, running fine. The ZED Unity plugin 3.8 cannot be loaded with DllNotFoundException. I have Ubuntu 20.04 and CUDA 11.7.
Pls. help me solve the problem!
Bests,

Mike

C-der-Baum, 2023-03-15:
Exact same error here with Unity 2021.3.21f1. ZED SDK 3.8.2 is installed, running fine. I have Ubuntu 18.04 and  CUDA 11.7.

C-der-Baum, 2023-03-15:
In Windows, it only shows a pink screen for Mono and crashes for stereo.

C-der-Baum, 2023-03-16:
OK, managed to get the video stream running on windows. I was choosing the universal rendering pipeline when starting a new project. When not doing that bit just some "empty" 2D project, it worked with the instructions as in the README. But now, the object detection does not work. The respective model has been optimized but it's not showing any boxes of people. Must I compile the OD scripts first?

closed 2023-04-22 by github-actions[bot]