stereolabs/zed-unity #246: [Unity 2022.1.23f] Zed2i - Skeleton Detection scene freezing/not launching

MlleDR, 2023-03-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

All the other example scenes seems to work fine except the SkeletonDetection scene, which, when launched in Unity, freezes Unity and forces me to kill the process. 

### Steps to Reproduce

1. Using Unity 2022.1.23f
2. Open the scene Assets/ZED/Examples/SkeletonTracking/Scene/SkeletonDemo.unity
3. On the ZED Manager, tried Depth mode to PERFORMANCE & ULTRA.
4. Input => USB
5. The rest of the manager's parameters have been left to default.
6. Launch the scene in game mode
7. >> Loading is infinite and the app will not respond until being killed in the Task Manager.

### Expected Result

For the scene to launch.

### Actual Result

Unity freezes and the scene is never launched.

### ZED Camera model

ZED2i

### Environment

```shell
Zed Diagnosis:
{
    "AI Models": {
        "info": [
            "MULTI CLASS DETECTION is not optimized",
            "MULTI CLASS MEDIUM DETECTION is not optimized",
            "HUMAN BODY FAST DETECTION is not optimized",
            "HUMAN BODY MEDIUM DETECTION is not optimized",
            "HUMAN BODY ACCURATE DETECTION is not optimized",
            "PERSON HEAD DETECTION is not optimized",
            "PERSON HEAD ACCURATE DETECTION is not optimized",
            "REID ASSOCIATION is not optimized"
        ],
        "valid": [
            "MULTI CLASS ACCURATE DETECTION is optimized",
            "NEURAL DEPTH is optimized"
        ]
    },
    "Camera Test": {
        "InternalDevicesCount": 2,
        "ZEDCount": 1,
        "camera": {
            "Device ID": 1,
            "Firmware": 1523,
            "Init Output": "OK",
            "Model": "ZED 2i",
            "Serial Number": 35127040,
            "valid": [
                "<b>Camera: </b>ZED 2i",
                "<b>Firmware: </b>1523"
            ]
        },
        "resolutions": {
            "HD1080@15": {
                "fps (input)": 15,
                "fps (output)": 15,
[... 434 lines cut ...]
            "sl_ai64.dll",
            "sl_zed64.dll",
            "zlibwapi.dll"
        ],
        "driverVersion": "528.49",
        "drivervDllVersion": "31.0.15.2849",
        "resourcesFiles": [
            "neural_depth_1.4.model",
            "neural_depth_1.4.model_optimized-aaabw-1-86-11070-8401-8402-13-192-2304-48-100-512-28-0-0e49",
            "objects_accurate_2.2.model",
            "objects_accurate_2.2.model_optimized-fbcbw-1-86-11070-8401-8402-13-192-2304-48-100-512-28-0",
            "skeleton_accurate_2.0.model",
            "skeleton_performance_2.1.model"
        ],
        "resourcesFilesExpected": [
            "objects_performance_2.2"
        ],
        "runtimeDllVersion": 11070,
        "valid": [
            "<b>ZED SDK version:</b> 3.8.2",
            "<b>CUDA version:</b> V11.7.64"
        ]
    }
}
```

Bvallon-sl, 2023-03-13:
Hi,

Thanks for reaching us out.

At the start of this scene, the ZED SDK checks if the selected AI detection model is downloaded and optimized on your machine.
If, not it will do it. In Unity, this process is not very well handled (I'll change that soon) as it is done in the main thread causing the Editor to freeze until the optimization is finished. As it can take few minutes to do, I'll recommend to manually optimize the AI models you wish to use outside of Unity, following this documentation (https://support.stereolabs.com/hc/en-us/articles/9747407795223-How-can-I-optimize-the-ZED-SDK-AI-models-manually-).

Best regards,
Benjamin Vallon

MlleDR, 2023-03-14:
Hi, thank you for your response.

All AI models are still downloading but here is a first error that I get :

[image]

Ingrid

Bvallon-sl, 2023-03-14:
Can you make sure you have access to internet while running the process ?

MlleDR, 2023-03-14:
Yes I did have access to internet while the process was in progress.

Bvallon-sl, 2023-03-14:
Can you re run the ZED Diagnostic tool and share the report and/or try to run a sample using one of the AI models, for example, "Body Tracking" ? 

I think the download failed the first time, tried again and succeeded the second time without saying anything.

MlleDR, 2023-03-26:
Hi

> Hi, thank you for your response.
> 
> The downloading process has ended with the following issues : [image]
> 
> Diagnostics [image]
> 
> Testing the same scene in Unity => The scene is still freezing the app, leaving a black screen and not loading any skeleton.
> 
> Ingrid

Hi, sorry for the late answer. Before your last message, I used another PC and started over all installations. The result from cmd is the same.

[image]

Could it be that the skeleton data's path have changed on your side?
Thanks

MlleDR, 2023-03-26:
> Can you re run the ZED Diagnostic tool and share the report and/or try to run a sample using one of the AI models, for example, "Body Tracking" ?
> 
> I think the download failed the first time, tried again and succeeded the second time without saying anything.

Also, what's the scene "Body Tracking" in Unity?

Bvallon-sl, 2023-03-28:
Hi,

I meant "Skeleton Tracking", I'm sorry.

According to the diagnostic tool, the AI models are downloaded and optimized.

Even if you had this error message during the optimization process, can you try to run any sample that uses the AI module to see if it works or not ?

closed 2023-05-04 by github-actions[bot]