stereolabs/zed-unity #262: dllz_retrieve_bodies_data run crash

CattleDrinkRedWine, 2023-07-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Unity version 2022.3.0f1c1
zed sdk version 4.0.3

zed diagnostic run well.no problem.The crash occurs when another person enters Zed's field of view.
[image]

### Steps to Reproduce

1.
2.
3.
...
i have no idea to reproduce this problem.

### Expected Result

crash

### Actual Result

Unity version 2022.3.0f1c1
zed sdk version 4.0.3

zed diagnostic run well.no problem.The crash occurs when another person enters Zed's field of view.

### ZED Camera model

ZED

### Environment

```shell
window 10
Unity version 2022.3.0f1c1
zed sdk version 4.0.3
```

SLJLMacarit, 2023-07-19:
Hi @CattleDrinkRedWine, thanks for the report. What release of the Unity plugin are you using?
Please update to 4.0.5 and check if you still have the issue.
Also, can you try a body tracking sample from the SDK and check if you have any issue at all?

You say you don't know how to reproduce it, do you reproduce it on your side? Does it always happen? What is the process to make it happen? Please list what you do, even if it's the "normal" usage of the plugin. What level are you using, does it crash with other levels...

Thanks in advance for this

CattleDrinkRedWine, 2023-07-19:
When I was still using version 4.0.1, I already reported this issue and you told me to upgrade to 4.0.3. Now you're asking me to upgrade to 4.0.5. Every time I receive a reply, it's the same. Is this the way Zed resolves issues? Is this always the attitude towards problem-solving?

***@***.***
 
From: SLJLMacarit
Date: 2023-07-19 16:27
To: stereolabs/zed-unity
CC: CattleDrinkRedWine; Mention
Subject: Re: [stereolabs/zed-unity] dllz_retrieve_bodies_data run crash (Issue #262)
Hi @CattleDrinkRedWine, thanks for the report. What release of the Unity plugin are you using?
Please update to 4.0.5 and check if you still have the issue.
Also, can you try a body tracking sample from the SDK and check if you have any issue at all?
You say you don't know how to reproduce it, do you reproduce it on your side? Does it always happen? What is the process to make it happen? Please list what you do, even if it's the "normal" usage of the plugin. What level are you using, does it crash with other levels...
Thanks in advance for this

SLJLMacarit, 2023-07-19:
@CattleDrinkRedWine 
That's fair, but please understand that the ZED SDK is still in "Early access". Thus, some development still happens, and I like to think this Unity plugin we provide is not small in size, so some problems may rise and be fixed along the versions. I am suggesting to update to be on the same version that we are doing our active testing on. This will facilitate the investigations on our side a lot, and accelerate the arrival of a fix by that much.

I understand the frustration of facing an unexpected crash, especially since it seems you encountered it before, but please understand that every issue reported is examined. If it's not fixed, we didn't have the time to look into it yet, because of either a lack of information on how to reproduce it, or a mis-prioritization, or a very unique issue difficult to address, etc.

That being said, please do not ignore the rest of the message, and provide a little more information, since I have a similar configuration to you and don't encounter this issue.

> Also, can you try a [body tracking sample from the SDK](https://github.com/stereolabs/zed-sdk/tree/master/body%20tracking) and check if you have any issue at all?
>
> You say you don't know how to [...]

closed 2023-08-24 by github-actions[bot]