stereolabs/zed-unity #265: Unity Crashes when I import Package

zacharyyamaoka, 2023-08-14:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When I import the ZED unity package, I first get an error regarding no xr mangement etc.

I install the XR mangement plugin and then some errors appear in the console and then the project crashes!

Errors seem related to ZEDRenderingPlane.cs - 

one says: AspectRatio is obsolete: AspectRatio enum has been deprecated and its functionality has been removed
[](url)
[image]

### Steps to Reproduce

Create new URP project in Unity 22.3.6f1
Import ZED Unity Package
Install XR Mangement

### Expected Result

Project to stay open and to be able to run the samples

### Actual Result

Unity Crashes

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Windows
SDK: 4.0
GPU: 1060
CUDA: 12
```

### Anything else?

Your help with this matter will be greatly appreicated!!!

Bvallon-sl, 2023-08-14:
Hi,

What you see here are warnings, not errors, that's not what is causing the crash.

Can you tell me what SDK and unity plugin versions are you using please?

zacharyyamaoka, 2023-08-14:
I am using the most recent ZED SDK from the downloads page:

CUDA 12
[ZED SDK for Windows 10/11](https://download.stereolabs.com/zedsdk/4.0/cu121/win)4.0

- The ZED explorer and depth viewer worked

I am using the [Zed_Unity_Plugin_v4.0.5.unitypackage](https://github.com/stereolabs/zed-unity/releases/download/v4.0.5/Zed_Unity_Plugin_v4.0.5.unitypackage)

as the the most recent .unitypackage was not avaliable, I will try agian now with the newest unity package,

Thank you so much for the fast reply! 

Zach

zacharyyamaoka, 2023-08-14:
When I create a URP template project, and run a scene with just the ZED_Rig_Mono, the game screen is pink.

When I create a HDRP template project, and run a scene with just the ZED_Rig_Mono, then the camera shows.

I was planning to use URP as I wasn't planning on doing any hyper realistic graphics, please advise! thank you : )

samphilips, 2023-08-15:
I am facing the same issue, the unity crashes when I try to install the XR plugin

zacharyyamaoka, 2023-08-15:
> I am facing the same issue, the unity crashes when I try to install the XR plugin , I am using 4.0.5 plugin, windows 11 , cuda 12.1

@samphilips try with the new 4.0.6 plugin!

closed 2023-09-24 by github-actions[bot]