stereolabs/zed-unity #266: Missing scripts ( prefabs missing scripts) - Body tracking module

samphilips, 2023-08-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I was trying to use the body tracking module, and was following the steps given in the ZED Unity documentation https://www.stereolabs.com/docs/unity/   , was facing the error that the prefabs are missing scripts, and the mono and stereo camera are missing scripts, images attached, I am currently using windows 11, with unity 2022.x , please advice what is the issue 

[image]

[image]

### Steps to Reproduce

1. import asset - unity zed plugin 
2. add camera - mono or stereo 

### Expected Result

1. get the zed manager script 

### Actual Result

[image]

### ZED Camera model

ZED2

### Environment

```shell
Windows 11 
gpu - nvidia RTX 3060 X 2 
CUDA Version : 12.1
ZED SDK 4.0 
Unity plugin : 4.0.5
Unity version 2022.x
```

SLJLMacarit, 2023-08-18:
Hi @samphilips,

Sorry for the delay.
Can you try adding the XR Management package from Project Settings -> XR Plugin Management?

samphilips, 2023-08-18:
> Hi @samphilips,
> 
> Sorry for the delay. Can you try adding the XR Management package from Project Settings -> XR Plugin Management?
> 
> Edit: Just saw your other comment on the crash with 4.0.6. I'll try some things and be back.

Thank you for your response, I was able to downgrade my system to windows 10 and try and it works for now, I am facing another error of camera not being found in the unity zed, set up, however, it works in the ZED SDK, Image attached but the unity isn't able to detect the camera, I have followed all the steps given in this documentation  https://www.stereolabs.com/docs/unity/body-tracking/#run-the-scene 

[image]

[image]

samphilips, 2023-08-18:
> > Hi @samphilips,
> > Sorry for the delay. Can you try adding the XR Management package from Project Settings -> XR Plugin Management?
> > Edit: Just saw your other comment on the crash with 4.0.6. I'll try some things and be back.
> 
> Thank you for your response @SLJLMacarit , I was able to downgrade my system to windows 10 and try and it works for now, I am facing another error of camera not being found in the unity zed, set up, however, it works in the ZED SDK, Image attached but the unity isn't able to detect the camera, I have followed all the steps given in this documentation https://www.stereolabs.com/docs/unity/body-tracking/#run-the-scene
> 
> [image]
> 
> ![image](https://user-images.githubusercontent.com/139512584/261690333-73923279-b66c-4d56-8301-6e2485eeb622.png

to give a broader perspective, this is how my unity screen looks like 
[image]

samphilips, 2023-08-21:
hi @SLJLMacarit could you please help me with this, what can be done to make this work, thank you

SLJLMacarit, 2023-08-22:
I'm happy it's solved by upgrading, but I'm sad because it didn't used to be necessary.
I logged this issue internally, we'll investigate it more in-depth.

closed 2023-08-22 by SLJLMacarit