stereolabs/zed-unity #269: Optimization process Timeout even after optimzing the AI models outside of Unity, using the ZED Diagnostic tool

aaronlsmiles, 2023-09-07:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Optimization process times out even after optimzing the AI models outside of Unity, using the ZED Diagnostic tool, as instructed in the [literature](https://support.stereolabs.com/hc/en-us/articles/9747407795223-How-can-I-optimize-the-ZED-SDK-AI-models-manually-). 

I have uninstalled and reinstalled, trying different versions of Unity (2021, 2022, etc.) and CUDA (11.8, 12.1, etc.), but the issue is always the same and the ZED Diagnostic tool shows everything with a green tick, except the AI Models, which show as unoptimised. 

I also can't find the optimised A models in any folder (e.g., C:\Program Files (x86)\ZED SDK, or C:\ProgramData\Stereolabs\resources (which is empty)).

### Steps to Reproduce

Followed the steps [here](https://www.stereolabs.com/docs/unity/object-detection/) to the letter.

### Expected Result

For the camera to initialise and the object detection model be working with 3D bounding boxes appearing on detected objects.

### Actual Result

Error: Optimization process Timeout. Please try to optimze the AI models outside of Unity, using the ZED Diagnostic tool 
UnityEngine.Debug:LogError (object)
ZEDManager/<startObjectDetection>d__379:MoveNext () (at Assets/ZED/SDK/Helpers/Scripts/ZEDManager.cs:3037)
UnityEngine.SetupCoroutine:InvokeMoveNext (System.Collections.IEnumerator,intptr)

### ZED Camera model

ZED Mini

### Environment

```shell
{
    "AI Models": {
        "info": [
            "MULTI CLASS DETECTION is not optimized",
            "MULTI CLASS MEDIUM DETECTION is not optimized",
            "MULTI CLASS ACCURATE DETECTION is not optimized",
            "HUMAN BODY FAST DETECTION is not optimized",
            "HUMAN BODY MEDIUM DETECTION is not optimized",
            "HUMAN BODY ACCURATE DETECTION is not optimized",
            "HUMAN BODY 38 FAST DETECTION is not optimized",
            "HUMAN BODY 38 MEDIUM DETECTION is not optimized",
            "HUMAN BODY 38 ACCURATE DETECTION is not optimized",
            "PERSON HEAD DETECTION is not optimized",
            "PERSON HEAD ACCURATE DETECTION is not optimized",
            "REID ASSOCIATION is not optimized",
            "NEURAL DEPTH is not optimized"
        ]
    },
    "Camera Test": {
        "InternalDevicesCount": 2,
        "ZEDCount": 1,
        "camera": {
            "Device ID": 1,
            "Firmware": 1523,
            "Init Output": "OK",
            "Model": "ZED Mini",
            "Serial Number": 16396036,
            "valid": [
                "<b>Camera: </b>ZED Mini",
                "<b>Firmware: </b>1523"
            ]
        },
        "resolutions": {
            "HD1080@15": {
                "fps (input)": 15,
[... 357 lines cut ...]
            "nvinfer_dispatch.dll",
            "nvinfer_lean.dll",
            "nvinfer_plugin.dll",
            "nvinfer_vc_plugin.dll",
            "nvonnxparser.dll",
            "nvparsers.dll",
            "sl_ai64.dll",
            "sl_zed64.dll",
            "zlibwapi.dll"
        ],
        "driverVersion": "531.61",
        "drivervDllVersion": "31.0.15.3161",
        "resourcesFiles": [
        ],
        "resourcesFilesExpected": [
            "objects_performance_3.2"
        ],
        "runtimeDllVersion": 12010,
        "valid": [
            "<b>ZED SDK version:</b> 4.0.6",
            "<b>CUDA version:</b> V12.1.66"
        ]
    }
}
```

aaronlsmiles, 2023-09-10:
This should be possible with ZED Mini as it states [here](https://www.stereolabs.com/docs/object-detection/):

> Important: At the moment, only a few object classes can be detected and tracked with the 3D Object Detection API using ZED cameras (except the ZED 1 camera). 

Can anyone elaborate please, this is urgent!?

SLJLMacarit, 2023-09-12:
Hi @aaronlsmiles, sorry for the delay.
You should very much be able to optimize the models, at least outside of Unity.
Does optimizing the models outside of Unity work? You can find some directives here: [How can I optimize the ZED SDK AI models manually?](https://support.stereolabs.com/hc/en-us/articles/9747407795223-How-can-I-optimize-the-ZED-SDK-AI-models-manually-)

aaronlsmiles, 2023-09-25:
I contacted support and there was some issue with the optimised files not downloading as the resources folder was always empty. So they sent them across and I pasted them in the folder then optimised them, which worked. They are investigating the issue.

closed 2023-09-25 by aaronlsmiles