stereolabs/zed-unity #273: Direct3D12 not supported

keli95566, 2023-10-06:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

All current SDKs do not support direct3D12, can't use the ZED sdk with projects with D3D12 requirements. When will this graphics backend be supported?

SLJLMacarit, 2023-10-16:
Hi,

It is in our backlog but it's not requested much, so we didn't plan to implement it short-term.
We'll investigate the changes needed and evaluate the priority again.

keli95566, 2023-11-02:
Hi there, is the native plugin of ZED Unity based on the ZED C++ implementation? Would it be possible in the future to open source the native-plugin C++ project as well? This makes it easier to extend and implement any features on top of the existing plugins.

closed 2023-11-06 by SLJLMacarit