stereolabs/zed-unity #276: Thousands of "NullReferenceException: Object reference not set to an instance of an object" Errors

mericgeren, 2023-12-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello,

I am trying to set up and use the Zed Unity Plugin. For this end, I have imported v4.0.7 version of the plugin on a device in which ZED SDK v4.0.8, CUDA 12.2 and DirectX 12 installed (from NVIDIA Control Panel I can also see that Direct3D feature level is 12_1). To this end, I have created a project with 3D URP template. After that, I have imported the package as a custom package (I have selected Zed_Unity_Plugin_v4.0.7.unityproject file). Then, I have deleted the main camera and then, from package manager I have installed VR feature, XR Plugin Management and XR Interaction Toolkit packages. After installing packages, I have tested Pointcloud and Drone Shooter examples. When testing both examples for the first time, the device and an Oculus Quest 2 headset was connected via cable. On these tries, the Unity became unresponsive and just froze for several minutes while on the headset, nothing have opened until I have closed Unity from task manager. Later on, I have disconnected the VR headset and tried both samples again. On both scene, as soon as I have hit the play button I have welcomed by an error about Direct3D11 not being supported followed by another error reads: "NullReferenceException: Object reference not set to an instance of an object" repeted more than a thousand times. You can see the screenshots of errors I got below: 

[image]
[image]
[image]

### Steps to Reproduce

1. Create a unity project with 3D URP template
2. Import Zed Unity Package as custom package.
3. Install needed VR packages through package manager
4. Try to run the examples provided in the Zed Unity package while VR headset is connected.
5. Observe that both Unity and VR headset freezing
6. Disconnect the headset
7. Run examples while VR headset isn't connected.
8. Observe that there is a repeating error on console.
...

### Expected Result

Being able to see the streoscopic view provided by a ZED Mini camera through VR headset and/or screen of device I am using.

### Actual Result

I have seen more than a thousand of these error messages. telling me "NullReferenceException: Object reference not set to an instance of an object".

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Windows 11 22H2
CPU: 13th Gen Intel(R) Core(TM) i7-13650HX
GPU: NVIDA RTX4060
ZED SDK Version: 4.0.8
ZED Unity Plugin Version: 4.0.7
Other Information: Unity 21.3.32f1
```

SLJLMacarit, 2023-12-15:
Hello @mericgeren,
Thanks for reaching out, it seems a part of our documentation needs reworking, indeed. The part about [importing into a new URP project](https://www.stereolabs.com/docs/unity#create-a-new-urp-project) lacks a step.

You have to update the ZED package to URP after importing it.
Please click on **ZED -> Upgrade ZED Plugin to URP** in the top menu.

mericgeren, 2023-12-26:
Thank you for the information,

After taking that extra step as suggested, all exceptions have been cleared. If you don't mind I am cosing this issue as solved.

Kindest regards.

closed 2023-12-26 by mericgeren