stereolabs/zed-unity #284: The ZED plugin cannot be loaded

Nikolatesla-lj, 2024-03-13:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

[image]
[image]
[image]

### Steps to Reproduce

Follow the steps to install the file "Zed_Unity_Plugin_v4.0.7.unitypackage", the following error occurs

### Expected Result

unity3D is working properly

### Actual Result

DllNotFoundException
UnityEngine.Debug:Log (object)
sl.ZEDCamera:CheckPlugin () (at Assets/ZED/SDK/NativeInterface/ZEDCamera.cs:920)
ZEDPluginInspector:UpdateWnd () (at Assets/ZED/Editor/Scripts/ZEDPluginInspector.cs:159)
UnityEditor.EditorApplication:Internal_CallUpdateFunctions () (at /home/bokken/build/output/unity/unity/Editor/Mono/EditorApplication.cs:381)
————————————————————————————————-——————————

The ZED plugin cannot be loaded. 
 Please check that you have ZED SDK v4.0 installed

 If the problem persists, please contact our support team at support@stereolabs.com

UnityEngine.Debug:LogError (object)
sl.ZEDCamera:CheckPlugin () (at Assets/ZED/SDK/NativeInterface/ZEDCamera.cs:921)
ZEDPluginInspector:UpdateWnd () (at Assets/ZED/Editor/Scripts/ZEDPluginInspector.cs:159)
UnityEditor.EditorApplication:Internal_CallUpdateFunctions () (at /home/bokken/build/output/unity/unity/Editor/Mono/EditorApplication.cs:381)

### ZED Camera model

ZED2i

### Environment

```shell
OS：Ubuntu 22.04
unity 3D：2022.3.21.f1
ZED SDK： 4.0.8
CUDA:12.1

Zed_Unity_Plugin_v4.0.7
```

closed 2024-04-18 by github-actions[bot]

keith-lackey-steamroller-tech, 2024-09-12:
I'm running into this same exact issue. how do i resolve?

ethanguo770, 2025-01-17:
https://community.stereolabs.com/t/unity-error-zed-sdk-is-not-installed-or-needs-to-be-updated/7218/3