stereolabs/zed-unity #288: Point Cloud Demo not working in Unity 2022

MoritzSkowronski, 2024-04-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

In Unity 2022 in all render pipelines, the Point Cloud demo does not show any points on screen. The issue seems to be related to the shader, which still compiles, but draws no points. The point cloud fusion demo that uses a different approach still works in comparison.

### Steps to Reproduce

1. Install Unity 2022
2. Import Zed
3. Open Point Cloud Demo and run it

### Expected Result

A point cloud is rendered to the screen

### Actual Result

Nothing renders

### ZED Camera model

ZED2

### Environment

```shell
OS: Windows 11
GPU: Geforce RTX 3090
```

Shunichi09, 2024-05-06:
I encountered the same error.
My environment is

- Unity 2022.3.25f1
- Window11
- ZED 2i

Does anyone solve this problem?

SLJLMacarit, 2024-05-06:
@MoritzSkowronski @Shunichi09 

Thank you for the report, it seems it started happening along the 2022.5.
I reproduced the problem in 2022.5.27, not in 2022.5.5.

We'll look into this.

Shunichi09, 2024-05-07:
Thank you for your response! I appreciate your attention to this matter. Please let us know if you need any further information. Looking forward to hearing from you soon.

SLJLMacarit, 2024-05-13:
Hi there, 
This seems to work again in 2023.x. This could be a Unity problem, breaking during 2022 lifetime and fixed in 2023.

Sorry, as it only seems to break the point cloud sample "temporarily", we will not look further into this.

closed 2024-05-13 by SLJLMacarit