stereolabs/zed-unity #301: gitub link not working

jungwonshin, 2025-04-05:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When I add the package git url, the unity shuts down. Please fix this.

### Steps to Reproduce

1.
2.
3.
...

### Expected Result

It should be installed on Unity

### Actual Result

Unity Crashes

### ZED Camera model

ZED

### Environment

```shell
OS: windows
```

Bvallon-sl, 2025-04-07:
Hi,

Thanks for reporting. It seems the Unity project crashes at start if there is a version mismatch between the plugin and the SDK version installed on your machine. I'll investigate further this week and see what exactly is causing this.

Sorry for the inconvenience.

jungwonshin, 2025-04-08:
I tried new project and now the program doesn't crash but getting this error despite zed sdk being latest version.

[image]

Bvallon-sl, 2025-04-09:
It should be fixed now, can you confirm?

closed 2025-05-16 by github-actions[bot]